ilyakurdyukov/spreadtrum_flash #2: ver expected

medanisjbara, 2023-03-23:
I tried running this on [Energizer Energy E12](https://www.gsmarena.com/energizer_energy_e12-9684.php) (which has a Spreadtrum SC6531E according to the specifications). and I got `ver expected` as an output.
The output seems to be from [this ERR_EXIT](https://github.com/ilyakurdyukov/spreadtrum_flash/blob/main/spd_dump.c#L1072).
Does this mean that the chipset in the specification page is not the correct one ? 
If I understand correctly, this error message is printed if the data received is not in the correct format ?

I should note that I didn't try any other firmware dumper other than this. I will comment here if I ever did.

medanisjbara, 2023-03-23:
Note: I'm not pressing any boot key as I wasn't able to locate any key that makes a difference in the delay time before boot.

ilyakurdyukov, 2023-03-24:
This is what happens if you turn on without holding the boot key or haven't found the right one.

ilyakurdyukov, 2023-03-24:
> recv_type(io): 0xffffffff

Means timeout, the device is not responding. Which is normal if you haven't enabled boot mode, because you aren't holding the right boot key.

I can find the boot key from the firmware for this phone model, but I didn't find this firmware on the internet.

ilyakurdyukov, 2023-03-24:
Before looking for a boot key, check the chip model. Pull out the battery and connect to USB, if after a few seconds it shows that charging is in progress - this is SC6531E (look for a boot key without a battery), if not - it means SC6531DA (for which it is more difficult to look for a boot key, because you need to pull out the battery every time).

I found on the internet that some models require pressing two keys at the same time, but I did not come across such.

ilyakurdyukov, 2023-03-24:
You can use an OTG adapter or a special boot cable to bypass the boot key.

medanisjbara, 2023-03-24:
> Means timeout, the device is not responding.

Thank you for the explanation. Is there a chance you can add this to the README ?
> I didn't find this firmware on the internet.

Unfortunately, the fact that it's not available is why I am trying to dump it. However, while looking, I was able to find [this website](https://gsmusbdriver.com/energizer-energy-e12) which contains some USB drivers (and I'm assuming that they can be helpful in retrieving the firmware without having to know the boot key). I think I will look into that as a last resort.

ilyakurdyukov, 2023-03-24:
> which contains some USB drivers (and I'm assuming that they can be helpful in retrieving the firmware without having to know the boot key)

Drivers will not help you avoid the boot key, only a boot cable (or an OTG adapter) can help.

This site is telling you bullshit, because they have generic Spreadtrum drivers, not specific to this model. What you can see from the filenames.

medanisjbara, 2023-03-24:
> if after a few seconds it shows that charging is in progress

This doesn't seem to be the case. The screen flashes for a split second and gets black again before flashing again and so on. I'm assuming this is how SC6531DA behaves.
> for which it is more difficult to look for a boot key, because you need to pull out the battery every time

I did exactly that this time and was able to determine the boot key (which is the number "1").
```
BSL_REP_VER: "SPRD3\0"
loadfile("nor_fdl1.bin") failed
```
Awesome! Thank you.

ilyakurdyukov, 2023-03-24:
These drivers are for use with Spreadtrum's flashing tools, which will also require you to know the boot key or use a boot cable.

medanisjbara, 2023-03-24:
> You can use an OTG adapter or a special boot cable to bypass the boot key.

I would like to learn more about how this is done if possible.

closed 2023-03-24 by medanisjbara

ilyakurdyukov, 2023-03-24:
> This doesn't seem to be the case. The screen flashes for a split second and gets black again before flashing again and so on. I'm assuming this is how SC6531DA behaves.

I'm not sure about this, the SC6531DA just won't connect via USB when plugged in without a battery. I don't remember if the screen is flashing.

> ```
> BSL_REP_VER: "SPRD3\0"
> loadfile("nor_fdl1.bin") failed
> ```

You can find `nor_fdl1.bin` in [Releases](https://github.com/ilyakurdyukov/spreadtrum_flash/releases).

ilyakurdyukov, 2023-03-24:
> > You can use an OTG adapter or a special boot cable to bypass the boot key.
> 
> I would like to learn more about how this is done if possible.

This is explained in the Readme:

> Instead of finding the boot key (sometimes there's no boot key, as on smart watches with only the power key), it's more convenient to use a boot cable with shorted 4th and 5th pins. This is the same as for OTG adapters, so you can combine an OTG adapter with an AM to AM USB cable.

medanisjbara, 2023-03-24:
Okay, thank you.