ilyakurdyukov/spreadtrum_flash #3: repartition failed

TomKing062, 2023-06-21:
https://github.com/ilyakurdyukov/spreadtrum_flash/blob/6886a85e6e9d80a5e02d8c9afd1c991b20a36b35/spd_dump.c#LL826C2-L826C2

now the packet end with userdata size ff ff ff 7f (BSL_REP_OPERATION_FAILED, 0x84)

 add `if(size == 2147483647) size = -1;` before `WRITE32_LE(buf + 0x48, size);`

then the packet end with userdata size ff ff ff ff (repartition successfully)

TomKing062, 2023-06-21:
maybe it's cause by gcc(or system related)? hardcoded 2147483647 is not good, this just solved my own solution.

ilyakurdyukov, 2023-06-21:
Can you show the end of the partition table from the .xml?

```
                <Partition id="userdata" size="0xFFFFFFFF"/>
            </Partitions>
```

Did you compile on Windows or Linux?

TomKing062, 2023-06-21:
[image]
windows msys2 mingw64

ilyakurdyukov, 2023-06-21:
Okay, then it's the same problem as with the strtol() function on Windows.

ilyakurdyukov, 2023-06-21:
fixed: https://github.com/ilyakurdyukov/spreadtrum_flash/commit/fa20bcc314c931e0c790ba62ab028706d802ccc3

Windows saturates values read from scanf/strtol, while Linux truncates them.

closed 2023-06-21 by ilyakurdyukov