ilyakurdyukov/spreadtrum_flash #8: A problem

1702703423, 2024-02-24:
Hello,when i use this tools to get the system.img from a UD 710 devices , it reported: bad length<8191,expected 12294> on my computer,what should i do

ilyakurdyukov, 2024-02-24:
Try adding `disable_transcode` command before `read_part`.

ilyakurdyukov, 2024-02-24:
If that doesn't help, try decreasing `blk_size` to 0x1000.

1702703423, 2024-02-24:
after i add the disable_transcode,it now report timeout reach.i also change to the 0x1000 but it cant load the fdl1,may be i should find another pc to try ,thanks your help

ilyakurdyukov, 2024-02-24:
> after i add the disable_transcode,it now report timeout reach.i also change to the 0x1000 but it cant load the fdl1,may be i should find another pc to try ,thanks your help

It looks like you didn't turn off your device before trying again. This is why you can't load `fdl1` again.

CE1CECL, 2024-03-07:
> after i add the disable_transcode,it now report timeout reach.i also change to the 0x1000 but it cant load the fdl1,may be i should find another pc to try ,thanks your help

I had a similar problem in issue #9 can you share what command you are using with us @1702703423 (trying to help out)