ilyakurdyukov/spreadtrum_flash #10: Just a little question that I have

CrazyBoy2014, 2024-04-13:
Hi there!

I am trying to dump the firmware of an Alcatel 2053, which has the SC6531E.
So, when I tried to find the boot key, I tried the asterisk. The charging message wouldn't display until I stop holding. And it also didn't work.
Is this a sign that I'm getting very close to finding the boot key? My current assumption is that it's a combination with the asterisk and another button.

ilyakurdyukov, 2024-04-13:
I read about phones where you have to press two keys, but I didn't come across such models. It will be difficult to find such a combination. You can make a boot cable from a micro USB OTG adapter and a USB A male to A male cable. With this cable you don't need to know the boot key.

ilyakurdyukov, 2024-04-13:
Judging by the firmware it should just be an asterisk.

CrazyBoy2014, 2024-04-14:
Got it. Thanks!

closed 2024-04-14 by CrazyBoy2014