ilyakurdyukov/spreadtrum_flash #11: usb_send failed : LIBUSB_ERROR_TIMEOUT

Z9044234820, 2024-06-28:


ilyakurdyukov, 2024-06-28:
Describe what you are doing and under what circumstances the problem occurred, or I will close this issue.

Z9044234820, 2024-06-28:
When I use SPD_dump to dump the user partition to test the program, it uploads the FDL file successfully, but afterwards it shows this LIBUSB error (My device is SC9830)
```
$ sudo spd_dump         --verbose 2 --wait 1000000 fdl ./fdl1.bin 0x50000000 fdl ./fdl2.bin 0x9F000000 disable_transcode read_part user_partition 0 1M user_part.bin
[truncated]
send (536):
7e 00 02 02 10 00 00 00 00 00 00 00 00 00 00 00  |~...............|
00 00 00 00 00 03 00 00 00 04 00 00 00 02 00 00  |................|
00 01 00 00 00 0b 00 00 00 12 01 01 01 00 00 00  |................|
40 82 17 00 4d 02 02 00 00 00 01 00 00 09 02 20  |@...M...........|
00 01 01 00 c0 32 09 04 00 00 02 ff 00 00 00 07  |.....2..........|
05 85 02 40 00 00 07 05 06 02 40 00 00 53 70 72  |...@......@..Spr|
65 61 64 74 72 75 6d 20 42 6f 6f 74 20 42 6c 6f  |eadtrum.Boot.Blo|
63 6b 20 76 65 72 73 69 6f 6e 20 31 2e 31 00 46  |ck.version.1.1.F|
44 4c 5f 50 61 63 6b 65 74 44 6f 49 64 6c 65 00  |DL_PacketDoIdle.|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 c0 5f  |..............._|
81 00 00 01 03 02 06 03 09 04 0d 05 10 06 13 07  |................|
16 08 19 09 1c 0a 1f 0b 22 0c 26 0d 29 0e 2c 0f  |........".&.).,.|
2f 10 32 11 35 12 38 13 3b 14 3f 15 42 16 45 17  |/.2.5.8.;.?.B.E.|
48 18 4b 19 4e 1a 51 1b 54 1c 58 1d 5b 1e 5e 1f  |H.K.N.Q.T.X.[.^.|
61 aa aa 55 55 01 00 00 00 80 5d 77 50 00 a0 25  |a..UU.....]wP..%|
26 00 84 d7 17 00 c2 eb 0b 03 00 00 00 01 00 00  |&...............|
00 03 00 00 00 03 00 00 00 e8 03 00 00 84 03 00  |................|
00 00 00 00 00 34 08 00 00 f1 ff ff 3f 00 00 00  |.....4......?...|
00 00 00 00 00 00 00 00 00 55 55 aa aa 55 14 00  |.........UU..U..|
00 e1 15 00 00 02 19 00 00 23 1c 00 00 44 1f 00  |.........#...D..|
00 60 22 00 00 86 25 00 00 a7 28 00 00 0d 0a 64  |.`"...%...(....d|
64 72 20 74 72 61 69 6e 69 6e 67 20 6e 6f 74 20  |dr.training.not.|
64 6f 6e 65 21 00 0d 0a 64 64 72 20 74 72 61 69  |done!...ddr.trai|
6e 69 6e 67 20 65 72 72 6f 72 21 00 0d 0a 64 64  |ning.error!...dd|
72 20 61 75 74 6f 20 64 65 74 65 63 74 20 63 73  |r.auto.detect.cs|
30 20 66 61 69 6c 21 20 72 65 74 72 79 20 64 65  |0.fail!.retry.de|
74 65 63 74 00 0d 0a 64 64 72 20 61 75 74 6f 20  |tect...ddr.auto.|
64 65 74 65 63 74 20 63 73 30 20 66 61 69 6c 21  |detect.cs0.fail!|
00 0d 0a 64 64 72 20 61 75 74 6f 20 64 65 74 65  |...ddr.auto.dete|
63 74 20 63 73 30 20 6f 6b 21 00 0d 0a 64 64 72  |ct.cs0.ok!...ddr|
20 61 75 74 6f 20 64 65 74 65 63 74 20 63 73 31  |.auto.detect.cs1|
20 66 61 69 6c 21 00 0d 0a 20 63 61 6e 20 6e 6f  |.fail!....can.no|
74 20 66 69 6e 64 20 73 75 70 70 6f 72 74 65 64  |t.find.supported|
20 64 72 61 6d fe ae 7e                          |.dram..~|
recv (8):
7e 00 80 00 00 3b 5a 7e                          |~....;Z~|
send (536):
7e 00 02 02 10 20 69 6e 66 6f 21 00 54 68 65 20  |~.....info!.The.|
66 69 72 73 74 20 46 44 4c 20 66 61 69 6c 65 64  |first.FDL.failed|
21 0d 0a 00 64 61 74 61 5f 73 69 7a 65 20 65 72  |!...data_size.er|
72 6f 72 20 3a 20 64 61 74 61 73 69 7a 65 20 3d  |ror.:.datasize.=|
20 25 64 20 20 4d 41 58 5f 50 4b 54 5f 53 49 5a  |.%d..MAX_PKT_SIZ|
45 20 3d 20 25 64 0a 00 0a 0a 0a 25 73 20 25 73  |E.=.%d.....%s.%s|
20 25 64 0a 0a 0a 00 2f 68 6f 6d 65 2f 63 6d 65  |.%d..../home/cme|
2f 4c 44 30 31 33 30 2f 77 31 35 2e 34 33 2f 32  |/LD0130/w15.43/2|
30 31 36 30 37 32 38 2d 72 65 6c 65 61 73 65 2f  |0160728-release/|
73 70 72 64 5f 61 6e 64 72 6f 69 64 2f 63 68 69  |sprd_android/chi|
70 72 61 6d 2f 6e 61 6e 64 5f 66 64 6c 2f 63 6f  |pram/nand_fdl/co|
6d 6d 6f 6e 2f 70 61 63 6b 65 74 2e 63 00 25 34  |mmon/packet.c.%4|
78 25 34 78 00 25 32 78 00 0a 0a 0a 0a 00 ff ff  |x%4x.%2x........|
ff ff ff ff ff c0 00 00 00 09 0d 0a 0e 0f 04 03  |................|
00 0c 0a 06 03 00 a0 a0 00 06 03 04 05 03 03 02  |................|
02 05 00 02 02 02 02 00 00 02 02 03 03 1b 00 17  |................|
00 2b 00 17 00 01 00 26 00 00 00 00 00 05 06 00  |.+.....&........|
00 04 00 83 00 00 00 02 00 02 00 06 00 00 00 04  |................|
06 83 00 00 00 04 00 00 00 14 07 02 07 12 00 46  |...............F|
e0 bb db a0 00 c8 00 00 00 09 0d 0a 0e 0f 04 03  |................|
00 0c 0a 06 03 00 a0 a0 00 06 03 04 06 03 03 02  |................|
02 05 00 02 02 02 02 00 00 02 02 03 03 1c 00 18  |................|
00 2c 00 18 00 01 00 28 00 00 00 00 00 05 07 00  |.,.....(........|
00 04 00 83 00 00 00 02 00 02 00 07 00 00 00 04  |................|
06 83 00 00 00 04 00 00 00 14 07 02 07 12 00 48  |...............H|
e0 e2 e4 a0 00 80 01 00 00 11 1a 14 0f 1d 04 03  |................|
00 0c 0b 06 03 00 a0 a0 00 0c 05 04 0a 06 06 02  |................|
02 05 00 02 02 02 02 00 00 03 03 04 04 34 00 2e  |.............4..|
[... 134 lines cut ...]
70 00 c2 01 00 00 00 30 70 00 c2 01 00 00 00 10  |p......0p.......|
70 00 c2 01 00 00 00 00 70 00 c2 01 00 4b 2d 7e  |p.......p....K-~|
recv (8):
7e 00 80 00 00 3b 5a 7e                          |~....;Z~|
send (8):
7e 00 03 00 00 59 50 7e                          |~....YP~|
recv (8):
7e 00 80 00 00 3b 5a 7e                          |~....;Z~|
send (8):
7e 00 04 00 00 dc c0 7e                          |~......~|
recv (8):
7e 00 80 00 00 3b 5a 7e                          |~....;Z~|
checkbaud 1
send (4):
7e 7e 7e 7e                                      |~~~~|
checkbaud 2
send (4):
7e 7e 7e 7e                                      |~~~~|
checkbaud 3
send (4):
7e 7e 7e 7e                                      |~~~~|
checkbaud 4
send (4):
7e 7e 7e 7e                                      |~~~~|
usb_send failed : LIBUSB_ERROR_TIMEOUT

ilyakurdyukov, 2024-06-28:
FDL2 is not responding, possible reasons:

1. Wrong FDL2, for a different chip revision. If you don't have the original firmware and are using FDLs from smartphones on the same chip, then some FDLs may not work.

2. FDL2 is loaded to the wrong address.

There is nothing wrong with `libusb` as FDL1 works.

Z9044234820, 2024-06-28:
I do not have the original firmware, I am using FDL1 and FDL2 from different smartphone pac files with the same chip. I even tried without FDL2 but same error. Is FDL2 required for this to work?

ilyakurdyukov, 2024-06-28:
> I even tried without FDL2 but same error.

I mentioned FDL2, but it could also be FDL1. It seems to be FDL1 if it stops at `spd_dump fdl ./fdl1.bin 0x50000000`.

* `--wait N` is not a timeout (it is waiting N seconds for the device to connect), timeout is `timeout N`

ilyakurdyukov, 2024-06-28:
I found that there may be four revisions of this chip: SC9830 SC9830A SC9830I SC9830L

Z9044234820, 2024-06-29:
Could it also be due to wrong FDL 1 which results in the LIBUSB error timeout?

ilyakurdyukov, 2024-06-29:
Yes, I said that in https://github.com/ilyakurdyukov/spreadtrum_flash/issues/11#issuecomment-2196951345

Try to load FDL1 first, nothing else, if you see this error, it means FDL1 is hung.