ilyakurdyukov/spreadtrum_flash #14: How to make the baudrate higher?

alebalasov, 2024-09-06:
Is it possible to change the baudrate of the port to make the dumper work faster? I saw the parameter, but not shure how to change it correctly.

ilyakurdyukov, 2024-09-07:
Many FDLs ignore the CHANGE_BAUD command, most likely this command is only for UART connection. Some people forked my repo and [added](https://github.com/TomKing062/spreadtrum_flash/blob/fc1f333d8e0303a6ab9b80ac91588d89a1a8e756/spd_dump_interactive.c#L328) this command, but you may notice that this code is under `!USE_LIBUSB` condition.

alebalasov, 2024-09-07:
Thank. Than, the only possible way to make it for "non-libusb" is to send the parameter "disable_transcode" after fdl2 if it is supported by the model. Am I right?

ilyakurdyukov, 2024-09-07:
You can try sending this command on FDL1 stage to see if it does anything.

Tiger T310's FDL2 doesn't have a handler for this command, FDL1 has a handler for UART mode.

If you are using USB connection, it will remain USB connection. I don't know what you imagine transcoding to be, but it's just converting certain byte values ​​to an escape sequence and back:
```
7d -> 7d 5d
7e -> 7d 5e
```

KiMiGuel, 2026-08-17:
I actually have modded spd_dump for linux that fixes the CHECK_BAUD error if its useful, you can find it on my repo called [Root Guide] Cubot KingKong ES3 Unisoc T615