ilyakurdyukov/spreadtrum_flash #17: issue with boot

ZoriXL, 2024-10-27:
How to reboot phone to flas mode? I try all keys but they dont work. I want to boot with this program, its possible? Phone: Nokia TA-1203, Spreadtrum 6531e

ilyakurdyukov, 2024-10-28:
How are you sure it's Spreadtrum? I can't find any information about it except from very unreliable sources that could be wrong. Does the phone connect to the PC via USB under vendor ID 1782?

ilyakurdyukov, 2024-10-28:
Usually if you run `spd_dump` in advance, then connect the phone to USB, and if the boot key is not pressed, then `spd_dump` prints `ver expected`.

Also you can try holding down as many buttons at the same time as possible, sometimes it works. But don't hold the power button.

ZoriXL, 2024-10-28:
я попробую без кнопки питания, при подключении программа пишет Waiting for
connect (300) и нечего не происходит, драйвер стоит, телефон походу
загрузился с клавишей 7, у него черный экран и он не запускается, но почему
то spd_dump.exe его не видит

ilyakurdyukov, 2024-10-28:
> spd_dump.exe

Значит Windows, значит драйвер не установлен или неправильно установлен, насчёт Windows я проконсультировать не могу, читайте help-ы от libusb, там для каждой версии Windows всё по своему. Предпочтительно использовать Linux.

ZoriXL, 2024-10-28:
а как использовать Линукс если на гитхабе в релизе только exe

ilyakurdyukov, 2024-10-28:
> а как использовать Линукс если на гитхабе в релизе только exe

На Линкусе принято компилировать самому.

> Use `make`, `libusb/libusb-dev` packages must be installed.

ZoriXL, 2024-10-28:
понял, спасибо

ilyakurdyukov, 2024-10-29:
Я делал ссылку в Readme для пользователей Windows:

https://github.com/libusb/libusb/wiki/Windows#driver-installation

В Линукс драйвер не нужен, в Windows нужно устанавливать типовой драйвер для каждого конкретного идентификатора устройства, чтобы с ним работать через libusb. Для этого есть программа Zadig, запускаем её, подключаем телефон в загрузочном режиме, должно найти устройство `1782:4d00`, первая цифра это vendor ID, вторая это device ID. Если vendor ID совпадает, но другой device ID - значит устройство находится в другом режиме. Там разные варианты драйвера, для разных версий Windows, тут надо инструкции читать.

Вот пример дампа 4МБ флэш памяти:

`spd_dump.exe end_data 0 fdl nor_fdl1.bin 0x40004000 read_flash 0x80000003 0 4M flash.bin`

Для Нокий тут добавлено `end_data 0`, это для обхода проверки подписей FDL. Не нужно при использовании оригинальных FDL от похожих моделей.

Прошивка может быть и больше 4МБ, надо дампать больше и смотреть повторяется или нет, обрезать когда нанёт повторяться. Но для этой модели должно быть 4МБ, потому что это урезанная версия TA-1174, а там было 4МБ.