ilyakurdyukov/spreadtrum_flash #18: Does it support SC9832E?

DRCRecoveryData, 2024-10-28:
Hi,

How spd_dump get the list partition when in fdl1 and fdl2 was sent?

ilyakurdyukov, 2024-10-29:
I don't know, FDL API changes over time, some older chips may not support partition list command.

You can try dumping the GPT:

`read_part user_partition 0 1M gpt.bin`

DRCRecoveryData, 2024-10-29:
> I don't know, FDL API changes over time, some older chips may not support partition list command.
> 
> 
> 
> You can try dumping the GPT:
> 
> 
> 
> `read_part user_partition 0 1M gpt.bin`

so in gpt.img they have list partition name and list?

i have dump this via unlocktool

https://drive.google.com/file/d/102rXbA2zgD1vEO_idjyt_rwxcc33Olfu/view?usp=drivesdk

ilyakurdyukov, 2024-10-29:
Here is a description of the format:

https://en.wikipedia.org/wiki/GUID_Partition_Table

You may be interested in the fork of spreadtrum_flash: https://github.com/TomKing062/spreadtrum_flash

Which does it automatically, but is even worse documented than mine.

ilyakurdyukov, 2024-10-29:
> i have dump this via unlocktool

Looks like the format that FDL should return.

Have you even tried the `partition_list` command that described in the Readme?

`partition_list <partition.xml> - saves the current partition list in .xml format. Also prints as text.`

DRCRecoveryData, 2024-10-29:
> > i have dump this via unlocktool
> 
> 
> 
> Looks like the format that FDL should return.
> 
> 
> 
> Have you even tried the `partition_list` command that described in the Readme?
> 
> 
> 
> `partition_list <partition.xml> - saves the current partition list in .xml format. Also prints as text.`

i'll try it, does the ```repartition``` mean this use the gpt table file overwrite to current gpt on emmc of devive?

ilyakurdyukov, 2024-10-29:
The `repartition` command will use the list of partitions from the .xml, it is intended to use such a list from the .xml, that can be found in the unpacked firmware. Keep in mind that this is a destructive operation (partitions will not be moved if one partition in the middle is resized), you must know what you are doing.

```xml
            <Partitions>
                <!-- size unit is MBytes -->
                <Partition id="prodnv" size="10"/>
                <Partition id="miscdata" size="1"/>
                <Partition id="misc" size="1"/>
                <Partition id="trustos_a" size="6"/>
                <Partition id="trustos_b" size="6"/>
                <Partition id="sml_a" size="1"/>
                <Partition id="sml_b" size="1"/>
```

CE1CECL, 2025-01-20:
I think this is a duplicate of http://github.com/ilyakurdyukov/spreadtrum_flash/issues/6 (sc9832e is the exact same as sp9832e, the naming is just weird).
And yes, the 9832e is missing commands, the U-Boot most vendors ship is just as old as the SoC, even if they released the phone today.

DianeuF, 2025-05-24:
Судя по странице https://phonedb.net/index.php?m=processor&id=769&c=spreadtrum_sc9832e
чип SC9832E использует память формата eMMC 5.0 и команда partition_list не сможет прочитать список разделов.

CE1CECL, 2025-05-25:
> Судя по странице https://phonedb.net/index.php?m=processor&id=769&c=spreadtrum_sc9832e
> чип SC9832E использует память формата eMMC 5.0 и команда partition_list не сможет прочитать список разделов.

partition_list only works on some newer FDL2s, in my case, the latest u-boot grabbed from Google OTA for my C5L 2020 (FWVGA) it works, but not on the stock stock one from 2020. It also works when I compiled it from leaked source code, either by Unisoc, or BLU (BLU Gave me the U-Boot source for the C0070WW since UBoot is GPL)

CE1CECL, 2025-11-07:
> This is very weird. Every spreadtrum kernel I download, including ones with a matching 4.14.193 version, have the same `Error: selected processor does not support 'cpsid i' in ARM mode` message show up.
> 
> I assumed since the kernels were made for 64bit builds, they likely included spreadtrum Makefiles that broke 32bit compilation. But all their sprd-kernels? Even the Realme ones are messed up like this so it must originate from spreadtrum.
> 
> Maybe these kernels don't actually come from one common "internal spreadtrum branch" and rather are a bunch of forked off kernels..? Or maybe my build commands are wrong?!
> 
> To verify the former, aka each SoC has a different branch that breaks one-another, I'll track down [source for a device](https://download.ztedevices.com/device/global/support/opensource/2/20210327_01/ZTE%20Blade%20A31%20Lite_Android%2011(R%20GO)_kernel(4.14.193).tar.gz#W3) with the same kernel/SoC combo as this one.
> 
> For the latter, since you've properly compiled sprd-kernels before, please tell me if this looks sane:
> 
> `PATH="clang-r377782c/bin:gcc-4.9/bin:${PATH}" make -j1 ARCH=arm CC=clang LLVM=1 CLANG_TRIPLE=arm-linux-gnu- CROSS_COMPILE=arm-linux-androideabi- O=output` I've cross-compiled kernels before tho and never had issues (Qualcomm CAF 3.4, 3.10 & Mainline. Even patched them to work with the latest GCC and they still booted fine)
> ### Nevermind my build command was off. Switching to a gcc4.9-only compile fixes it.
> 
> But it should theoretically build with the same clang it was originally built with..? I'm guessing clang's probably using my system's assembler rather than the gcc4.9 one I'm trying to coax it into using. Too lazy to try and fix it. If gcc4.9's build gives me issues then maybe I will.
> 
> I guess that means that even my "clang-built" kernel that failed to boot earlier was really assembled by modern GCC.. And that included the hacky source patches I did to force away my build errors.
> 
> Rebuilding those same kernels, it still doesn't work. I'll try the C5L-2020 on this device tomorrow.

Back again, sorry about the wait, its been a hot minute!

Anyways, can I ask which image are you packing into the boot.img? Are you using `arch/arm/boot/Image` instead of zImage or other, at least for me I had to use Image. Also, can you maybe upload your vbmeta files, as I can't somehow find your exact stock rom for you model, C0171WW. It might not do much, as you already got magisk working it seems, but the C5L-2020-FWVGA's kernel wasn't building with your config in your image :(, though it did with the original C5L 2020 Config I have in there (for the original device)

RDS5, 2025-11-17:
> Yeah, AFAIK the bootloader is so picky you not only have to use Image, but you also have to use AIK-Linux, to keep the spl versions/dates intact, else it wont boot. Have you tried to unpack the magisk img file and repack it without kernel changes and see it that alone boots?

Yeah, just repacking it alone boots. I'm using the raw mkbootimg/unpackbootimg tools from Android's SDK (already have it on my machine, why not use it?).
iirc I kept the dates intact by passing through the props, even attempting to keep the same order in which they came.
It got to a point where I `xxd boot-customkernel.img > custom.hex && xxd boot-magisk.img > magisk.hex && meld magisk.hex custom.hex`

There were some mild differences in the header that I'm pretty sure I corrected. Still didn't boot..
I've got a feeling there might still be something different in my header but I'll have to go back into the hex editor to verify. 
My machine's doing a restore (fs corruption) so I can't right now (should be available by tomorrow)

> root@LenovoLegionT5:/home/remoteadmin/VBHelp# file ./RDS5/AIK-Linux/split_img/magisk-signed.img-kernel
> ./RDS5/AIK-Linux/split_img/magisk-signed.img-kernel: data
> root@LenovoLegionT5:/home/remoteadmin/VBHelp# 

> (zImage wouldn't say "data", it would have a proper header which the sprd bootloader doesn't somehow like)

Huh. Maybe the zImage is compiled as ELF or something?

RDS5, 2025-12-30:
> (should be available by tomorrow)

famous last words
half the data is corrupted because i stupidly thought ext4 had a good enough fsck to survive trimming like 10mb off the RAID disks. long story short, fsck crashed and oom'd.
i was admittedly embarassed from this and procrastinated trying to fix it because of how stupid of an error it was.
<sup>atleast i didn't lose my family's entire archive of childhood photos this time</sup>

Anyways, alongside my requests to BLU for their source code, I've tried to recreate the exact toolchain used for this device.
I've successfully compiled the kernel using clang r377782c with the only piece of gcc being 4.9's assembler afaict.
Still, it's throwing an unwieldy amount of errors which makes me feel like they likely used a different assembler on this kernel.
Tomorrow morning I'll attempt to boot it and, if it fails, ill try to look into what exactly is different in my toolchain.

Worst case scenario, ive already soldered and made holes for the UART headers of this board. Ill buy a level shifter if I still can't get anything booting.