ilyakurdyukov/spreadtrum_flash #22: Как скачать таблицу разделов диска с телефона?

DianeuF, 2025-04-12:
Мне подойдёт команда
```
./spd_dump --verbose 0 --wait 300 \
	keep_charge 1 \
	disable_transcode \
	partition_list partition.xml \
	blk_size 0x3000 \
	power_off
```
?

ilyakurdyukov, 2025-04-16:
Не подойдёт, читайте readme внимательно. Нужно найти правильные FDL1/FDL2 бинарники для конкретного чипа, из поставки оригинальной прошивки или прошивки для устройства с тем же чипом. И сначала загрузить FDL бинарники, а потом уже команды выполнять.

DianeuF, 2025-04-16:
Илья, спасибо, буду пробовать.

ilyakurdyukov, 2025-04-22:
Надо ставить `--verbose 2` и смотреть в конец лога, на какую команду FDL не ответил.

DianeuF, 2025-05-10:
Я попробовал запустить программу с записью вывода в текстовый файл
./spd_dump --verbose 2 --wait 300 keep_charge 1 fdl fdl1.bin 0x50000000 fdl fdl2.bin 0x9F000000 disable_transcode partition_list partition.xml blk_size 0x3000 read_part logo 0 8M logo.bmp power_off > log.txt 2>&1
Итоговый лог-файл доступен по ссылке: https://wdfiles.ru/2a3vi

ilyakurdyukov, 2025-05-13:
Достаточно было этого:

```
send (8):
7e 00 04 00 00 ff fb 7e                          |~......~|
recv (8):
7e 00 96 00 00 ff 69 7e                          |~.....i~|
FDL2: incompatible partition
send (8):
7e 00 21 00 00 ff de 7e                          |~.!....~|
timeout reached
```

Не отвечает на команду `disable_transcode`, что первая команда после FDL2:

1. FDL2 завис, возможно от другого смартфона и не совместим с вашим. Кроме самого чипа существует разница в плате. Не все FDL подходят, даже на устройство с тем же чипом. У меня были безобидные вещи, от неродного FDL включался фонарик.
2. Команда `disable_transcode` не поддерживается. Но в таком случае должно возвращать ошибку.
3. Долго думает. Можно увеличить таймаут. Добавить `timeout 60000` перед `disable_transcode`. 

Какой это чип?

ilyakurdyukov, 2025-05-14:
Старый чип, думаю там другой API для чтения в FDL.

DianeuF, 2025-05-15:
Попробовал более старую прошивку скачать , v12
И попробовал read_part вместо read_flash поскольку с точкой старта 0 программа вряд ли поймет какой из разделов ей нужно скачивать
И решил скачать prodnv.img, который есть в прошивке

> ./spd_dump --verbose 2 --wait 300 keep_charge 1 fdl ../v12/fdl1.bin 0x50000000 fdl ../v12/fdl2.bin 0x9F000000 blk_size 0x3000 read_part prodnv 0 5M prodnv.zip power_off > log-1.txt 2>&1

В итоге файл скачался правильного размера 5мб.

DianeuF, 2025-05-20:
А файлы boot.img, system.img, recovery.img скачались размером с раздел на диске. Их видимо нужно обрезать какой-то специальной программой.
Попробовал записать recovery.img из новой прошивки v14, но выдало ошибку fopen.
./spd_dump --verbose 0 --wait 300 keep_charge 1 fdl ../v14/fdl1.bin 0x50000000 fdl ../v14/fdl2.bin 0x9F000000 \
write_part recovery 0 20M ../v14/recovery.img \
power_off
Waiting for connection (300s)
BSL_REP_VER: "SPRD3\0"
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
FDL2: incompatible partition
fopen(load) failed

ilyakurdyukov, 2025-05-20:
`write_part <partition_name> <input_file>`

Надо убрать "0 20M", указание офсета не поддерживается API.

DianeuF, 2025-05-20:
По поводу udev rules, у меня не получилось воспользоваться вашим примером, и я чуть-чуть его изменил.
Мой пример работает для debian 12.
/etc/udev/rules.d/80-android.rules
```
#	Spreadtrum
SUBSYSTEM=="usb", ATTR{idVendor}=="1782", ATTR{idProduct}=="4d00", SYMLINK+="android_flash"  ENV{adb_user}="yes"
SUBSYSTEM=="usb", ATTR{idVendor}=="1782", ATTR{idProduct}=="4011", SYMLINK+="android_charge"  ENV{adb_user}="yes"
# Enable device as a user device if found
ENV{adb_user}=="yes", MODE="0666", GROUP="plugdev", TAG+="uaccess"
```

ilyakurdyukov, 2025-05-21:
Зачем было обновлять таким сложным образом, чтобы сохранить пользовательские данные?

ilyakurdyukov, 2025-05-22:
Так как `partition_list` этими FDL не поддерживаются, то проверьте поддерживается ли `user_partition` раздел, что по сути чтение всего флэша одним куском, и в начале должен быть GPT (GUID Partition Table). По крайней мере в современных смартфонах это GPT, а что в старых было - я не знаю.

`read_part user_partition 0 1M gpt.bin`

DianeuF, 2025-05-22:
Нет, этой командой не считывается диск, код ошибки 96. Наверное из-за отсутствия root на телефоне. Но для прошивки таблица разделов впринципе не нужна.
```
./spd_dump --verbose 0 --wait 300 keep_charge 1 fdl ../v14/fdl1.bin 0x50000000 fdl ../v14/fdl2.bin 0x9F000000 read_part user_partition 0 1M ./skach/user_partition.bin power_off
BSL_REP_VER: "SPRD3\0"
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
FDL2: incompatible partition
unexpected response (0x0096)
```

ilyakurdyukov, 2025-05-22:
Глянул исходники U-Boot на гитхабе (функция "common_raw_read" из sprd_common_rw.c), не у всех есть фича с "user_partition".

DianeuF, 2025-05-22:
При включении root на телефоне всё равно не скачивается user_partition. 
Видимо вы правы, в файле 
https://github.com/mr-dumpling/u-boot15/blob/master/common/sprd_common_rw.c
строка 317
`dev_desc = get_dev("ufs", dev_id);`
видимо означает, что нужен тип памяти UFS, а в SC9832 тип памяти eMMC 4.5

closed 2025-05-24 by DianeuF