ilyakurdyukov/spreadtrum_flash #24: How to dump multiple partitions without unplugging after USB disconnect?

wiktorpp, 2025-04-25:
Hi.

I'm trying to dump multiple partitions one after another using a single spreadtrum_flash session. Here's the command I'm running:

sudo ./spd_dump --verbose 0 --wait 300 \
    fdl fdl1-sign.bin 0x65000800 \
    fdl lk-fdl2-sign.bin 0x9EFFFE00 \
    read_part prodnv 0 64M prodnv.img \
    read_part miscdata 0 1M miscdata.img \
    read_part misc 0 1M misc.img \
    ... (more read_part commands) ...
    read_part super 0 5600M super.img \
    ... (more read_part commands) ...
    power_off

The dumping process starts successfully, and several partitions are read without issues, but at some point I encounter:

usb_send failed : LIBUSB_ERROR_NO_DEVICE

It seems like the device disconnects itself (maybe resets or crashes) during the long session.
Currently, I have to unplug and replug the USB cable to continue.

closed 2025-04-25 by wiktorpp

ilyakurdyukov, 2025-04-25:
Use `disable_transcode` command after loading fdl2, maybe then it will be more stable and faster.

ilyakurdyukov, 2025-04-25:
It could also be a watchdog timer. How long does it take before disconnecting? (use `time -p ./spd_dump ...`)

P.S.: Use udev rules to run the tool without sudo.