ilyakurdyukov/spreadtrum_flash #25: Problem with unpac, dealing with more than 4GB size

aleb4lasov, 2025-05-03:
The major problem with unpac tool is that the .pac firmware more than 4GB in size can`t be unpacked properly. All parts, except super img, looks fine, but super has only some megabytes. Can it be corrected some way? Thanks.

ilyakurdyukov, 2025-05-04:
Yes, file IO needs to be ported to 64 bit. Can you provide a link with an example archive larger than 4GB?

aleb4lasov, 2025-05-04:
Here you are
https://www.needrom.com/download/tecno-spark-10c-ki5m/?Tok=486844_6557d20391b46cf43b8fdd1f6c22e4b7

aleb4lasov, 2025-05-04:
And, just one more question, if you don't mind. Could you, please, tell me, what compression algorithm is used? I just want to repack my own firmware. Thanks in advance.

ilyakurdyukov, 2025-05-05:
There is no compression or encryption, but this is not a regular archive, each file contains a lot of metadata.

ilyakurdyukov, 2025-05-05:
fixed in https://github.com/ilyakurdyukov/spreadtrum_flash/commit/a08cd9e2acdcbe0e8a08ad0ba6a7755a4f1200df

closed 2025-05-05 by ilyakurdyukov