ilyakurdyukov/spreadtrum_flash #31: usb_send failed (0/1) on moto e13

DavidTheDev556, 2025-09-27:
after plugging it in and it shows "SPD U25 Diag", after a couple of seconds, the script aborts because usb_send failed and i also hear a disconnect sound when it fails, and i have confirmed the device is powered off when i do this

ilyakurdyukov, 2025-09-27:
What is shown and where, on the screen or in the console? When booting from bootloader mode, there should be nothing on the screen, nor any sounds.

What script? It's a native app, not a script.

DavidTheDev556, 2025-09-27:
after connecting the phone, pressing vol - and running the bat script, the program establishes a connection and then somehow disconnects, because the device tries to boot into the power off battery % mode. also, do i need to run adb reboot bootloader beforehand?

ilyakurdyukov, 2025-09-27:
Try vol+, it could be any key (except the power button).

DavidTheDev556, 2025-09-27:
vol+ works too but it still ignores the connection and boots into power off battery level mode. I have read that shorting the test pin is necessary? Do I really have to? And if I do, do I just press the point with a random object?

ilyakurdyukov, 2025-09-27:
The instructions [here](https://www.softwarecrackguru.com/2025/05/motorola-moto-e13-latest-version-frp.html) say that this is the volume down key.

ilyakurdyukov, 2025-09-27:
This is a new chip, so Unisoc could have changed something in bootloader mode.

DavidTheDev556, 2025-09-27:
so you're saying the only way is to try to short the test pin? After that what do I do? I don't know any unlocking tools other than this one

ilyakurdyukov, 2025-09-27:
Shorting 4 and 5 pins in the micro-USB cable, of using micro-USB OTG adapter. But this phone seems to be using Type-C.

DavidTheDev556, 2025-09-27:
The test pin on this phone is next to the camera, from what i read.

ilyakurdyukov, 2025-09-27:
I don't know such pins, boot key should work.

You can try Unisoc's ResearchDownload application, but it will repartition your device if you successfully connect, even if you just want to read the partitions.

ilyakurdyukov, 2025-09-27:
You can find the firmware update file for this phone, update it, and then use ResearchDownload with this firmware file. Repartitioning won't affect anything, as the partitions will be the same size. You'll then be able to read and write the partitions using ResearchDownload.

ilyakurdyukov, 2025-09-27:
Well, you need to understand Android partitions, then you might be able to unlock it. It'll be difficult to figure out if you've never done it before. Moreover, methods may change over time.

closed 2025-09-27 by DavidTheDev556